la1k/libpredict #21: Unit tests

bjorgan, 2015-05-15:
Need tests for checking correctness of libpredict. Given specific TLE and a set of timestamps, generate file(s) with properties to check from original predict. Write functions to test the developed library functions against the properties in this file. 

For AOS/LOS/moonrise/sunrise, can also run some kind of sanity check? Loop through the orbit, check if AOS and LOS actually happens, or passes are missed, and how accurate the results are. Call AOS/LOS repeatedly at different times to see whether the results fall within a specified threshold.

bjorgan, 2015-09-23:
We now have tests and (test data generation from predict) against: 

* Predicted orbit of satellites
* Predicted observation of satellites
* Predicted observation of sun
* Predicted observation of moon
* Sanity check of AOS/LOS for a given satellite (predict is not involved)

Are any more tests needed? 

Still remaining: Need to find satellites from which we can generate good test cases. SGP4, SDP4, geostationary, difficult orbits, ... .

ryeng, 2015-09-26:
Fixed by a45fd8caddcce7b2fd7ca1095ccbdd4a577d4d5e.

closed 2015-09-26 by ryeng