la1k/libpredict #31: Clean up predict_orbit_t

ryeng, 2015-10-01:
Some values seem to be duplicated, or maybe even triplicated by pointers to structs/character strings with the same information.

A clean separation between predict_tle_t and predict_orbit_t would help a lot. Currently, predict_tle_t is only used deep down in the library. It makes sense to parse a TLE string into predict_tle_t and then give the predict_tle_t as parameter to predict_create_orbit() instead of the string array it currently takes.

ryeng, 2015-10-01:
See also issue #22.

bjorgan, 2015-10-03:
Time-dependent, calculated orbit number will be called "revolution". The orbit number at epoch in the TLE-field should be called something like "revolution_at_epoch" in order to not cause confusion and to highlight the connection between the two.

commit 058de217 (2015-10-09): Merge pull request #39 from la1k/issue31

Closes #31.
closed 2015-10-09 by ryeng