la1k/libpredict #78: Issues compiling for ARM

bostrt, 2017-09-28:
I'm unable to compile for ARM and getting following error. If I rename the `s` variable, I can workaround but I'm not sure what the deal is. All works fine when compiling for x86.
```
# dnf install arm-none-eabi-gcc-cs-c++
# dnf install arm-none-eabi-newlib
# git clone https://github.com/la1k/libpredict.git
# git clone https://github.com/vpetrigo/arm-cmake-toolchains.git
# mkdir libpredict/build
# cd libpredict/build
# cmake -DCMAKE_TOOLCHAIN_FILE=../../arm-cmake-toolchains/arm-gcc-toolchain.cmake -DCMAKE_BUILD_TYPE=Debug ../
...
-- Build files have been written to: /tmp/libpredict/build
# make 
Scanning dependencies of target predict_static
[  2%] Building C object src/CMakeFiles/predict_static.dir/orbit.c.obj
In file included from /tmp/libpredict/src/orbit.c:4:0:
/tmp/libpredict/src/defs.h:24:12: error: expected identifier or '(' before numeric constant
 #define s  1.012229
            ^
make[2]: *** [src/CMakeFiles/predict_static.dir/build.make:63: src/CMakeFiles/predict_static.dir/orbit.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:124: src/CMakeFiles/predict_static.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

bjorgan, 2017-09-28:
Thanks for the issue! I had the same problem when trying to compile in Cygwin on Windows some months ago, but forgot about it. Turns out it can be reproduced by trying to compile a single file containing

```
#include "defs.h"
#include <time.h>
```

, but that it compiles fine without <time.h> included. 

If the contents of time.h is copied to a local file, and

```
#include "defs.h"
#include "time.h"
```

is compiled instead, the error message becomes a bit more meaningful:

```
$ gcc ../src/test.c
In file included from ../src/test.c:1:0:
../src/defs.h:24:12: error: expected identifier or ‘(’ before numeric constant
 #define s  1.012229
            ^
../src/time.h:111:7: note: in expansion of macro ‘s’
   int s;
       ^
```

The variable s in `struct __tzrule_struct` is expanded by the `s` define in defs.h. Seems like the only solution is to rename. The constant was already renamed in a different branch (https://github.com/la1k/libpredict/tree/issue58-deobfuscate-constants), which we'll try to review and merge soon. In the meantime, you could try to clone this branch or rename manually. Sorry about the inconvenience. :-)

bjorgan, 2017-11-25:
The misbehaving define was renamed in 61c46b0526a1e7cf9fd66a7af038a9367e224484.

closed 2017-11-25 by bjorgan