la1k/libpredict #83: Tracking of Solar System Objects

core-dd, 2017-11-21:
Allowing the user to track objects from the solar system in addition to satellites described by TLE files would allow a wider usage range of flyby. Objects coming to my mind are the following:

- [ ] Moon - Application Earth-Moon-Earth
- [ ] Jupiter - Observations of Noise Bursts (e.g. as calibration source)
- [ ] Sun - Observations of the Sun - probably for Hot Source/Cold Source Noise measurements
- [ ] Venus - Application Earth-Venus-Earth (there are Hams doing that ;)

core-dd, 2017-11-21:
`void predict_observe_moon(const predict_observer_t *observer, predict_julian_date_t time, struct predict_observation *obs);`
`void predict_observe_sun(const predict_observer_t *observer, predict_julian_date_t time, struct predict_observation *obs);`
are already implemented by `predict` legacy code.

ryeng, 2017-11-25:
Renamed issue to reflect which functionality is still missing.

core-dd, 2017-11-28:
Ongoing development in https://github.com/la1k/libpredict/tree/issue83-solar-system-objects and full VSOP87 based solution works to a certain precision (cbf6442ba4512d073104fae705d6f58ebe9620b0) but coefficients to calculate the osculating orbits blow the library size up. Looking into reduced precision calculations as pointing accuracy of most rotators is most likely not getting close to 0.1° and the opening angle of main lobe for a reasonably sized antenna system will be above 0.1° too.

core-dd, 2017-12-11:
I've ended up with using coefficients from [Simon et al.](http://adsabs.harvard.edu/abs/1994A%26A...282..663S) to calculate the mean elements of Venus and Jupiter (95c17437773b1dead0b7fa1b71426156ae8c7293), after some playing with VSOP87 and VSOP2013 based solutions. The VSOP87 solution was discarded as it requires to many coefficients to be included the library and the promised precision is unneeded for for the libpredict usecase. The current mean elements based solution gives a precision better than 0.5° and 0.6° for Venus and Jupiter, respectively. The comparison was done against data generated by the [Horizons](https://ssd.jpl.nasa.gov/horizons.cgi) tool for roughly each month in the timeframe from 2017 to 2025 and ended up as a testcase (587d2a8ea7b1be5ead3c9e146ece315a1890814d).
On the way I was using the equations in Meeus chapter 25 to rewrite the sun_predict() function (51e1156281652bb4854eada5580f56d10e90cf95) which might help to reduce the usage of some GPL'ed code.
Would be great to have some code-review to see if the current state of the code can be merged into master. :)