la1k/libpredict #114: Build fails on testcase_reader.cpp.o

horsaen, 2023-07-10:
```
[ 62%] Building CXX object tests/CMakeFiles/orbit-t.dir/testcase_reader.cpp.o
/home/horsaen/git/libpredict/tests/testcase_reader.cpp: In function ‘std::string& ltrim(std::string&)’:
/home/horsaen/git/libpredict/tests/testcase_reader.cpp:35:93: error: ‘std::pointer_to_unary_function<_Arg, _Result> std::ptr_fun(_Result (*)(_Arg)) [with _Arg = int; _Result = int]’ is deprecated: use 'std::function' instead [-Werror=deprecated-declarations]
   35 |         s.erase(s.begin(), std::find_if(s.begin(), s.end(), std::not1(std::ptr_fun<int, int>(std::isspace))));
      |                                                                       ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
In file included from /usr/include/c++/13.1.1/string:49,
                 from /home/horsaen/git/libpredict/tests/testcase_reader.h:4,
                 from /home/horsaen/git/libpredict/tests/testcase_reader.cpp:1:
/usr/include/c++/13.1.1/bits/stl_function.h:1126:5: note: declared here
 1126 |     ptr_fun(_Result (*__x)(_Arg))
      |     ^~~~~~~
/home/horsaen/git/libpredict/tests/testcase_reader.cpp:35:70: error: ‘constexpr std::unary_negate<_Predicate> std::not1(const _Predicate&) [with _Predicate = pointer_to_unary_function<int, int>]’ is deprecated: use 'std::not_fn' instead [-Werror=deprecated-declarations]
   35 |         s.erase(s.begin(), std::find_if(s.begin(), s.end(), std::not1(std::ptr_fun<int, int>(std::isspace))));
      |                                                             ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13.1.1/bits/stl_function.h:1046:5: note: declared here
 1046 |     not1(const _Predicate& __pred)
      |     ^~~~
/home/horsaen/git/libpredict/tests/testcase_reader.cpp: In function ‘std::string& rtrim(std::string&)’:
/home/horsaen/git/libpredict/tests/testcase_reader.cpp:41:84: error: ‘std::pointer_to_unary_function<_Arg, _Result> std::ptr_fun(_Result (*)(_Arg)) [with _Arg = int; _Result = int]’ is deprecated: use 'std::function' instead [-Werror=deprecated-declarations]
   41 |         s.erase(std::find_if(s.rbegin(), s.rend(), std::not1(std::ptr_fun<int, int>(std::isspace))).base(), s.end());
      |                                                              ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
/usr/include/c++/13.1.1/bits/stl_function.h:1126:5: note: declared here
 1126 |     ptr_fun(_Result (*__x)(_Arg))
      |     ^~~~~~~
/home/horsaen/git/libpredict/tests/testcase_reader.cpp:41:61: error: ‘constexpr std::unary_negate<_Predicate> std::not1(const _Predicate&) [with _Predicate = pointer_to_unary_function<int, int>]’ is deprecated: use 'std::not_fn' instead [-Werror=deprecated-declarations]
   41 |         s.erase(std::find_if(s.rbegin(), s.rend(), std::not1(std::ptr_fun<int, int>(std::isspace))).base(), s.end());
      |                                                    ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13.1.1/bits/stl_function.h:1046:5: note: declared here
 1046 |     not1(const _Predicate& __pred)
      |     ^~~~
cc1plus: all warnings being treated as errors
make[2]: *** [tests/CMakeFiles/orbit-t.dir/build.make:90: tests/CMakeFiles/orbit-t.dir/testcase_reader.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:314: tests/CMakeFiles/orbit-t.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

this occurs on arch linux, linux mint compiles fine.

any ideas ?

ploeffler, 2023-09-21:
I confirm that on Armbian

1-Bart-1, 2025-04-01:
This is still a problem. Either the `-Werror` in these two lines should be removed, or the deprecated code has to be updated.
https://github.com/la1k/libpredict/blob/47569d4e14b8fe55fabe39127936ee47c1613f96/CMakeLists.txt#L30-L31